makerbase-mks/MKS-DLC32 #98: Links for MKS-DLC32 firmware ERROR 404

daven78, 2022-01-07:
The CoreXY laser firmware link doesn't work.  I would like to download it for MKS-DLC32 board.

Ondrashek, 2022-01-07:
You can download it: 
https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware

daven78, 2022-01-07:
Where is it?  The blue link that says "For Laser, CoreXY machine" doesn't work.

[cid:1116d16a-9bb7-4cda-a20b-fcf97ab5bee3]

https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmakerbase-mks%2FMKS-DLC32%2Ftree%2Fmain%2Ffirmware&data=04%7C01%7C%7Cfa952fdb6c75489c1dd408d9d21626c6%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637771814855130529%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=QnHT1CUWV554gmrmb39ZkSfanWJKPmg47olKwvB7muk%3D&reserved=0>

Ondrashek, 2022-01-07:
> Where is it? The blue link that says "For Laser, CoreXY machine" doesn't work. [cid:1116d16a-9bb7-4cda-a20b-fcf97ab5bee3]
> […](#)

Yes, the blue link in description (readme) have bad url address. I sent you link to firmware tree: https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware

Select screen version TS24 or TS35 (if you don't have screen from makerbase, you can use one of them -> Laser -> CoreXY -> Board_v1.0 (for board version 1.x - jumpers) or Board_v2.0 (for board version 2.x - red dip switched) -> choose firmware by version.

closed 2022-02-25 by abinoer