makerbase-mks/MKS-DLC32 #99: DLC32 V2.0 Board Laser PWM and Spindle output together

Watto2021, 2022-01-10:
Hi, I have the above board using on a diode laser machine. I have everything working for laser side of things. I wish to add Air Assist to laser and require a 12V valve to turn air ON and OFF, I try to connect to spindle output but the voltage is low 3V~
Can I change $31 1000 will this give 12V at spindle? but not effect Laser PWM?

Ondrashek, 2022-01-10:
$31 with 1000 will be always 5V max in TTL laser pin

RandomHacks-Git, 2022-01-10:
The TTL and the spindle outputs are controlled by the same pin, you can't control them individually. 
The correct pin for air assist is SCL but you will need an external mosfet or a ttl signal relay to control your 12v valve.

Have a look here https://github.com/makerbase-mks/MKS-DLC32/issues/53

Watto2021, 2022-01-11:
Ok thank you Ondrashek and RandomHacks-Git. My multimeter was showing the spindle output 0-12V was following the TTL 0-5V output, im still learning the computer/program side so I assumed these were linked to same Processor pin.
RandomHacks thanks I was planning to use M8 M9 for the air assist as that matches my Plasma Table and SHEETCAM processor output, I just not sure which this pin was the output, does the SCL pin pull to ground with M8 command? wiring a mosfet to SCL for my 12V valve is in my knowledge (just) most microcontrollers i have experience pull the output pin to ground are these same?

Watto2021, 2022-01-11:
I have a spare of 3d printer heated bed mosfet, I assume i can connect the control cable to SCL but the second wire to 5V or GND? 
Will save me going to buy a std mosfet

[image]

makerbase-mks, 2022-01-11:
> I have a spare of 3d printer heated bed mosfet, I assume i can connect the control cable to SCL but the second wire to 5V or GND? Will save me going to buy a std mosfet
> 
> [image]

SCL->MOSFET signal in
GND->MOSFET GND
12V/24V->MOSFET 12/24V

Watto2021, 2022-01-14:
I have installed a mosfet as per wiring above. I have added an M7 and M8 into my G code after M3, and the file will not engrave.
The DLC32 V2.0 will have the file in the SD menu, when i select the file it will do the Edge/Bounding but then when I press engrave it goes to the 0% bar graph and displays PAUSE and STOP and never progress.

I have try and the DLC32 WILL still engrave from Wifi App
But WILL NOT engrave from any other G Code
I engraved from App then downloaded the G code Changed 1 X-axis position on PC then reload file, DLC32 Will not engrave once saved through another G code Writer

I Updated Firmware from DLC32_V2.0_D2.0.02_H24_公版_1107_N.bin
to
 V2.0.8_H24_20220107_N.bin
at the same time as doing this fitting if that makes a difference?

I will revert to the old firmware and test while I wait for a responce

Watto2021, 2022-01-15:
Thank you all,
I have all working correctly, My problem was with my G-code.
DLC32 Firmware will only execute M3, NOT M03. Also do not have a space in front of M commands can cause a Freeze.
Also DLC32 will Freeze .nc program if using M3 and M7 on the same Code Line, funny I have M5 and M30 on same Code Line and no issue.

closed 2022-01-20 by Watto2021