makerbase-mks/MKS-DLC32 #100: Power Module Footprint and schematic

NicNatTech, 2022-01-11:
I question whether the footprint for the motor drivers on page 7 of the wiring diagram, as well as its schematic are correct?
My first board blew the driver on socket # (Z) JUST on power up.
Please confirm

MKS-hosiey, 2022-01-12:
Did you install the driver backwards? Confirm the wiring diagram is correct

NicNatTech, 2022-01-12:
Per the Wiring Document  
[image]
Per Standard and schematic
[image]
Notice difference in pins 11-14
To clarify the only board I received blew two drivers on the z port on power up with no motors connected.

If I recall it had a black x on the underneath side of the port. (I will look tonight)
I have a replacement and others boards on order. (I believe this module is leaps and bounds ahead of the competition) 
Steps taken with board 
1) update firmware for CNC and  
2) install three motor drivers (XYZ)  (why did board only come with two drivers? 
3) power on with no motors connected( Z port blew two drivers X&Y no issues). 

I went looking at github documents and noticed the discrepancy.

MKS-hosiey, 2022-01-14:
The 1A ,1B and 2A and 2B ，will only lead to different direction, which will not cause burning driven. 

 

 
 
 
  

power on with no motors connected( Z port blew two drivers X&amp;Y no issues).

closed 2022-01-16 by NicNatTech