makerbase-mks/MKS-DLC32 #101: MKS-DLC32 and Y2 Axis with TMC2209

AngleV, 2022-01-15:
Hello

Recently i bought the MKS-DLC32 v.2.1 and the TMC2209 stepper drivers.
My goal is to connect 2 stepper motors to the Y Axis but the must work opposite.
Is this possible and if yes how can i do it properly?
Also does the TMC2209 will handle the current of the 2 stepper motors on it ?

Best regards
AngleV

manuelchaves, 2022-01-15:
Hi, I use 2 motors with no problems. Other solution is use external driver driver

AngleV, 2022-01-15:
@manuelchaves so you just plugged in the motors to the ports Y- 1 and Y-2 and they work opposite ??

Watto2021, 2022-01-16:
Yes, they should, the Y1 and Y2 have the A & B coils swapped on the circuit board layout. If both motors run the same direction when you plug in, you can just swap the A & B wires on just 1 of the Y steppers to fix it.

rizwanchishti, 2022-02-11:
Anyone knows the DIP configuration having DLC32 v2.1 and TMC 2209. Mine is working with M1 low and M2 high however the speed of the steppers is extremely slow. I was better on the eleksmaker mana board

AngleV, 2022-02-21:
@rizwanchishti 

Read this 

https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf

closed 2022-02-25 by abinoer

Shaileshunawane, 2023-02-28:
> Hi, I use 2 motors with no problems. Other solution is use external driver driver

how to use external drivers on this board. because it has only three points for external drivers and I want to use 4 stepper drivers( y1 and y2) so how to connect?

Shaileshunawane, 2023-02-28:
> Yes, they should, the Y1 and Y2 have the A & B coils swapped on the circuit board layout. If both motors run the same direction when you plug in, you can just swap the A & B wires on just 1 of the Y steppers to fix it.

how to use external drivers on this board. because it has only three points for external drivers and I want to use 4 stepper drivers( y1 and y2) so how to connect?