makerbase-mks/MKS-DLC32 #102: Lightburn inverse controls

Yamilli31, 2022-01-15:
Using lasergrbl the controls are perfect using lightburn the controls are upside down, any solution?

MKS-hosiey, 2022-01-17:
Is the motor moving in the opposite direction?

Yamilli31, 2022-01-17:
The movement of the engines on the Mks application and on the browser are inverse to lightburns

torenich, 2022-01-18:
Have you tried deleting and re-adding the GRBL device in Lightburn? I have a DLC32 v2 that I'm bench-testing. I had Lightburn installed and previously configured using different control boards. When I plugged the DLC32 in I got some really weird anomalies when using the arrows in the "Move" tab of Lightburn. Actually both left and right made the X motor spin the same direction!  After removing the previous GRBL device, then re-adding by letting Lightburn "Find My Laser" button, it automatically recognized the parameters of the DLC32 and everything worked after that.

Yamilli31, 2022-01-21:
The solution to the problem is the starting point on Lightburn that I had configured at the top left so that the program reverses the movement of the jog.

closed 2022-01-21 by Yamilli31