makerbase-mks/MKS-DLC32 #103: GCODE M4 DONT WORK

Cortapalos, 2022-01-15:
when the gcode file contains the command M4 (Dynamic power laser), the PWM laser output is not activated, with M3 (constant power) it works

Watto2021, 2022-01-16:
I did not try M4. Although I had lots of problems getting my laser to fire while also doing M7 for air assist. Fixed now.
I found M3 S 1-1000 worked fine to vary Laser Output power.
I also mention I found the  DLC32 would not work If I put more than 1 M command per line. So when I wanted M7 coolant On and M3 laser On. whichever was first on the line would work the second command would fail.
my G code post needed to be setup this way...
post.text (M3 S1000\n)
post.text (M7\n)

So the M commands are on a new line each

abinoer, 2022-01-17:
@Cortapalos I have tried to use M4 and it works normally. Which firmware do you use? Is it convenient to send out your Gcore file

dcbo, 2022-01-22:
> @Cortapalos I have tried to use M4 and it works normally. Which firmware do you use? Is it convenient to send out your Gcore file

For me M4 also does'nt work.

Board Version: MKS DLC32 CNC V2.0
Firmware: V2.08(8M.H24.20211223)

**Lightburn**
`[MSG:M4` requires laser mode or a reversable spindle]`

**WebGui:**
`M4
ERROR:20
error:20
`
Maybe it is a Setting Issue?

**My Setting**
`$$
$0=10
$1=5
$2=0
$3=7
$4=0
$5=1
$6=1
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=7
$24=300.000
$25=1000.000
$26=250.000
$27=1.000
$30=10000.000
$31=0.000
$32=1
$38=1
$37=1
$40=1
$100=80.000
$101=80.000
$102=80.000
$103=100.000
$104=100.000
$105=100.000
$110=6000.000
$111=6000.000
$112=6000.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=500.000
$121=500.000
$122=500.000
$123=200.000
$124=200.000
$125=200.000
$130=160.000
$131=160.000
$132=80.000
$133=300.000
$134=300.000
$135=300.000
ok
`

abinoer, 2022-01-24:
@dcbo You should use laser firmware

3d-wunschdruck, 2022-03-08:
Hi,
for me its the same message.. i have the laser version..
Can you please share the config.h file from grbl?
Maybe i can find the difference to my configuration..

abinoer, 2022-03-10:
The M4 works, but it only seems to work while on the move.This has nothing to do with parameter configuration

3d-wunschdruck, 2022-03-10:
That is correct, i have some artifacts if i‘m using the firmware below. I have compiled myself the grblhal and i‘m getting better results with the same gcode.
But my biggest Problem with the original firmware is the artifact in the raster engraving..
The picture on top is with grblhal and the other with the original firmware, in the face of naruto (below image) you see artifacts..

[image]

i would be very happy, if i can get the config.h
Or that you compile the grbl 1.1f version..

thank you very much..

abinoer, 2022-03-15:
@3d-wunschdruck Which firmware are you using? Do you use LightBurn? Can you upload your pictures here?

3d-wunsch, 2022-03-15:
[image]
I'm using V2.0.8_H24_20220107_N.bin Laser Firmware and lightburn as software.
I think i have solved this issue.
In grblHAL i have had missallignment in bidirectional scanning in lightburn.
So i gave the original Firmware a try.

I played around with the acceleration and ended in good results. I have set my acceleration higher and doesn`t have the artifacts like the second naruto.
Now it is working as it should. 
Only one thing that i have saw is, if i'm raster engraving, the speed drops if the laser is turned on. it like slower if it engraves.. in grblHAL it is almost the same speed, if i set it to 100mm/s then it is 100mm/s but with the mks firmware it is slowing down..
Anyway, it is working and i'm happy

abinoer, 2022-03-15:
Ok，thank you very much

closed 2022-03-16 by abinoer

stemaDev, 2022-12-01:
> For me M4 also does'nt work.
> ...
> **WebGui:** `M4 ERROR:20 error:20 ` Maybe it is a Setting Issue?

So the same thing for me. M3 works, but the movement is jerky, stutters a lot and is very slow. I get the same result from sd card and usb. When I use M4, the movement is fast and smooth, but the laser doesn't turn on. I know it only has to turn on when moving, but it does move and the laser still doesn't fire. Then I noticed the same "error: 20" that I get for an "M4 S10" command. I have $32=1, so laser mode is turned on. I'm not sure about my firmware version, I got the board a few months back and I'm not comfortable to install some chinese tool on my pc. Is there a way to put a .BIN file on the sd card so that it updates itself (like skr boards)? @dcbo, were you able to fix this somehow?

stemaDev, 2022-12-01:
> @dcbo You should use laser firmware

Where do I find that?

Flinner, 2024-03-10:
for all who still face this issue:

Use a laser firmware from https://github.com/makerbase-mks/MKS-DLC32/tree/main/MKS-DLC32-main/firmware/TS35/Laser/Normal/Board_V2.0

btw, V2.10_H35_2022_0621_N_ZX_001.bin has both Z axis and M4 instructions, related #333.

If you can't flash using the flash tool, use esptool.py on linux. I used the following command:

```
❯ esptool.py --baud 115200 --after no_reset write_flash --flash_size 8MB --flash_mode dout 0x00 V2.10_H35_2022_0621_N_ZX_001.bin
```

good luck all.