makerbase-mks/MKS-DLC32 #108: Printable box including LCD mount

capi92, 2022-01-21:
No issue, just sharing my MKS DLC32 box and 2.4 lcd mount...

https://www.prusaprinters.org/prints/93079-mks-dlc32-box-for-24-lcd-on-20x20-v-slot

manuelchaves, 2022-01-21:
Tks

pjrezai, 2022-01-23:
Hi! Could you remix it for the 3.5 inch lcd version? The "TS35"

capi92, 2022-01-23:
> Hi! Could you remix it for the 3.5 inch lcd version? The "TS35"

I don’t have the TS35 this hard for me to prototype it. 
the only way would be for someone to send be one…

kdaffy, 2022-01-24:
[image]

kdaffy, 2022-01-24:
FOR TS 35

kdaffy, 2022-01-24:
https://drive.google.com/file/d/1JM0WkJAN4q5972Bi0-t47BVEaOrr4bSw/view?usp=sharing

kdaffy, 2022-01-24:
https://drive.google.com/file/d/13MPnu7NrKNWzkU8K9PjkjpfworrrZjYh/view?usp=sharing

cryoglobe, 2022-05-12:
> FOR TS 35

Nice .. do you have a picture of the final "look" :) ?