makerbase-mks/MKS-DLC32 #111: MKS DLC32 resets / short circuits anytime both limit switches are reached

pjrezai, 2022-01-23:
Hi! I am a newbie, but I don't know why the board resets / short circuits anytime both limit switches are reached. Any help would be very appreciated! :)

makerbase-mks, 2022-01-23:
Maybe you should invert the level of the endstops

pjrezai, 2022-01-23:
> Maybe you should invert the level of the endstops

Thanks for the feedback!! Unfortunately my electronics knowledge is near 0. How would I do that? Thanks!!

pjrezai, 2022-01-24:
Is there any pictures of video I could see to make sure i'm installing it correctly so it doesn't short circuit? What do you mean by inverting them?

abinoer, 2022-01-25:
@pjrezai Short circuit? I think your wires are connected backwards. If you are using a mechanical limit, do not connect the 5V on the limit interface to avoid a short circuit caused by the wrong connection

pjrezai, 2022-01-31:
> @pjrezai Short circuit? I think your wires are connected backwards. If you are using a mechanical limit, do not connect the 5V on the limit interface to avoid a short circuit caused by the wrong connection

How do I not connect the 5V? Which one is that wire? Since all of them are connected by a pin connector

pjrezai, 2022-01-31:
Here are a couple pictures
[image]
[image]

pjrezai, 2022-02-04:
Any help would be appreciated! :)

abinoer, 2022-02-09:
There is screen printing on the motherboard, the red line on your picture is connected to 5V, you can pick out the red line

closed 2022-03-16 by abinoer