makerbase-mks/MKS-DLC32 #113: Tmc 2209

kdaffy, 2022-01-24:
Hi. Can I use tmc 2209 drivers with my board? Do i have to do something or I just take out the a4988 and put the new ones?thanks

Plottmania, 2022-01-24:
No problem, I use TMC 2208 and TMC 2209

kdaffy, 2022-01-24:
I ask this because i have read somewhere that you have to swap the 2209 in comparison with the 4988.

Plottmania, 2022-01-24:
I have among other things a Sculpfun with a MKS DLC32 and there run at the DLC32 V1.0 my TMC 2208 and at the DLC32 V2.0 my TMC 2209
When plugging on the right direction pay attention!
[image]

Devil-67, 2022-02-24:
which are the right ones ?
there are 2209 V1.2 V2.0 V3 V3.1

torenich, 2022-02-24:
> which are the right ones ?

I dont think it matters what hardware rev, but you must use the step/dir, also known as legacy, or stand alone version. You cannot use the SPI or UART versions.

kdaffy, 2022-03-02:
Hi, yesterday i received my tcm2209 from makerbase shop. I changed my old a4988 and changed the extended settingd $160, $161 and $162 from 16 to 256 (microsteps). They are very quiet but during an engraving suddenly it stopped engraving and i couldn't control the laser so i turned it off and on again. Does someone knows if when i change drivers i have to make any other settings? Thanks!!!

Devil-67, 2022-03-02:
which 2209 did you take ?
There are different versions 1,2 - 3,0 ?

closed 2022-08-10 by abinoer