makerbase-mks/MKS-DLC32 #114: J3 - Switch

Plottmania, 2022-01-24:
A small question...
What is the Switch J3 for?

[image]

THX

kdaffy, 2022-01-24:
you use it if you want to add a switch to the board to switch it on and off. if altough do this you have to remove the red fuse beside the switc
[image]
h

deltazerorsan, 2022-01-24:
It's an external power switch, but you must remove de fuse... And put one
also external.

abinoer, 2022-01-25:
@Plottmania You can read the manual to understand the functions of various [interfaces[](url)][https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf](url)

closed 2022-01-25 by abinoer