makerbase-mks/MKS-DLC32 #120: Probably the movement of the axes when controlled from the screen.

FGAzi, 2022-01-28:
Hello.  When working from LightBurn, when moving manually along the Y axis, the axis moves in the correct direction.  When moving the arrows from the screen, this axis moves in the opposite direction.  The other axis moves in the same way as from the LightBurn program.  How to make the movements when controlled from the program and the screen be in the same direction?

abinoer, 2022-02-08:
@FGAzi  Please update the latest firmware

Devil-67, 2022-02-20:
I have the firmware 2.08 H35

And the Y axis moves in the opposite direction in Lightburn, in LightBurn when Y+ is pressed it moves to Y-.
No matter how I change the config in Lightburn it is always one direction the wrong way round.

On the display of the MKS the Y axis is correct when Y+ then it also moves in Y+.

FGAzi, 2022-02-22:
@abinoer
> @FGAzi Please update the latest firmware

Tried using different firmware versions:
V2.0.7_H35_20211213_N
V2.0.8_H35_20211222_N
V2.0.8_H35_20220105_N. 
The problem is not solved.
Machine zero - upper left corner.

abinoer, 2022-02-23:
@FGAzi Note the Settings for LightBur
[image]
n

Devil-67, 2022-02-23:
Right, it's the same with me

abinoer, 2022-02-24:
You need to adjust the mechanical limit position

FGAzi, 2022-02-24:
@abinoer It is structurally impossible to change the position of the limit switches on the machine. This will create problems loading material onto the machine.
Can this be done programmatically or via settings?

GadgetX, 2022-02-24:
> @abinoer If change this setting, the axes will move correctly. In this case, the laser will move outside the working field of the program, since the mechanical limit switches are located in the upper left corner, and not in the lower left corner.

Did you try change  $23🤔

FGAzi, 2022-02-24:
@abinoer Changed parameters $3 and $23. Moves are not synced. Only changing the starting point helps.
When engraving from an Android application, the image is mirrored about the "Y" axis. This may be due to the different y-axes in LightBern and the Android app. Is it possible to add to the firmware, for example, settings for mirroring axes?