makerbase-mks/MKS-DLC32 #123: Z Axis issue

serper, 2022-02-01:
I have a problem, I don't know if it could be that the controller doesn't work well or the firmware (latest version). The Z axis motor sometimes works fine and sometimes it seems as if it gets stuck or goes in the opposite direction than it should and when changing directions it gets stuck again. I have changed the motor and cables and the same thing happens. If I put the cable from any other axis on the Z motor it works fine and the cable from the Z motor on any other axis fails as well. I changed the driver and it does the same thing.
It's totally random and I can't find a way to fix it.
Can you help me?

torenich, 2022-02-03:
I would check the current setting on your Z driver. It sounds like it is set to low.

serper, 2022-02-03:
Thanks a lot! I have turned a quarter of a turn to the right (to test) and I think it has been solved. I'll keep watching it. Now I will look at how to adjust it well by measuring the vref. I've spent hours with trial and error and it turns out that it was the simplest thing.
Many thanks!
Still I see that more people have something with the Z axis, maybe it is designed differently from the rest of the axes that makes it work slightly differently.

closed 2022-02-19 by abinoer