makerbase-mks/MKS-DLC32 #124: Z axis steper driver is HOT

koszokradziej, 2022-02-02:
I have problem with Z axis in my laser. Motor doesn't move and the steper driver is very hot eaven if the motor is unplug. I replaced the steper driver with a another one but the problem is the same (steper driver is OK). So i think the problem is in the motherboard.

Can you help me?

torenich, 2022-02-03:
You might want to check the current setting on your drivers. It sounds like it is set to high.

koszokradziej, 2022-02-03:
Every driver set on the same value (0,7V on resistor) but problem are only with z axis

makerbase-mks, 2022-02-05:
@koszokradziej Can you provide a image of your connection of the board?

solawc, 2022-02-08:
@koszokradziej After you connect the Z axis, is the Z axis always in a locked state? And when the X/Y axis is unlocked, is it still locked?

koszokradziej, 2022-02-08:
@solawc Yes

abinoer, 2023-01-09:
Try to change to a good motor drive, if there is still this problem, there may be hardware damage

closed 2023-01-09 by abinoer