makerbase-mks/MKS-DLC32 #125: Config.h and open source loading of our own firmware

Bobertsawesome, 2022-02-03:
I see that this project is still private with firmware, but that poses some issues that are vital to the function of the machine. For example, I want to be able to run Z-axis homing during normal conditions with the simple $H command. This is only configurable in the source config.h code of Grbl.

Are there any plans to make these functions editable through the laser tool? Or do we still have to wait for the firmware to become open source?

Thanks.

torenich, 2022-02-03:
Most of the settings you can change from the console with the extended settings commands. If you want to home Z along with X and Y, you could issue the

$Homing/Cycle0=XYZ <------homes all axes simultaneously

command to set them to home all at once. Or, a better practice for CNC, would have you home Z  first (up to clear work area), then home X and Y. You would issue these commands....

$Homing/Cycle0=Z <----homes Z first
$Homing/Cycle1=XY <---- then homes X and Y simultaneously _after_  Z has homed

These settings take effect and are saved as soon as they are issued. Then the next time (and every subsequent time) you issue the $H command, they with home in the order you set.

You can have 6 "homing cycles" 0-5, which will home your axes in that order.

You can also home each axes individually by using $HX, $HY, or $HZ.

Bobertsawesome, 2022-02-03:
> 

You, my sir, are a genius. I looked all over the GRBL Wiki and didn't see this.

closed 2022-02-03 by Bobertsawesome

daxliniere, 2022-04-09:
Thanks @torenich! Do you know if there is a list of the extended settings?

torenich, 2022-04-10:
Here is a link to the setting wiki from the original developer of Grbl_ESP32.

https://github.com/bdring/Grbl_Esp32/wiki/Settings

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE