makerbase-mks/MKS-DLC32 #127: Customize start screen

koszokradziej, 2022-02-05:
Is it possible to change the start screen to a personalized one, as it was possible to do in the MKS TFT35?

makerbase-mks, 2022-02-05:
Yes, we can modify for you. You can send the image.

koszokradziej, 2022-02-05:
In the attachment I am sending the start screen file (480/320 pixels) in the extension jpg and bin. I would ask you to put it in the MKS-DLC32 / firmware / TS35 / Laser / Normal / Board_V2.0 and MKS-DLC32 / firmware / TS35 / CNC / Normal / Board_v2.0. Thank you in advance
[Start Screen.zip](https://github.com/makerbase-mks/MKS-DLC32/files/8007940/Start.Screen.zip)

PEACE-CNC, 2022-02-17:
Hello, can I have this screen too?

johnrodrit, 2022-03-03:
I would like to have my screen, I will send you the image in the same way, I appreciate if you can do it, I use the laser firmwar
[logo.zip](https://github.com/makerbase-mks/MKS-DLC32/files/8180857/logo.zip)
e

LuisCnMan, 2022-05-01:
> Sí, podemos modificar por usted. Puede enviar la imagen.

Me pueden apoyar a cambiar la pantalla de inicio de la placa mks dlc 32

manuelchaves, 2022-05-02:
Hi, wouldn't it be more interesting for MKS to release the user himself to generate and u~pload your Start screen? Using IMAGE2LCD software, generating .BIN

dejanmaksa, 2022-05-29:
Hello, I also need a custom start screen. Here is my logo
[image]
I am using this firmware https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/Laser/Normal/Board_V2.0/V2.0.8_H35_20220105_N.bin

LuisCnMan, 2022-06-05:
> Sí, podemos modificar por usted. Puede enviar la imagen.

Pueden apoyarme a personalizar mi pantalla de inicio 

[image]

Medina-ui, 2022-06-13:
Hi. Is it possible to use the TFT35 available  firmware for DLC32 with the TFT35-FW? And is it possible to customize all images? not just the start screen as we can when using MKSTOOL for the previous versions?

PaoloViper74, 2022-10-22:
Hello,
Is it possible to customize the initial image? if so, in what way?

PaoloViper74, 2022-11-08:
Hello,
Is it possible to customize the initial image? if so, in what way?

Thanks

IsaqueLD, 2023-01-14:
Ola gostaria tambem de adicionar a Logo em Minhas CNCs que fabrico com o software de voces. 
[image]

3D-N00B, 2023-03-20:
Hi !
@makerbase-mks: how do I change the the start screen ?
Thank you

ocaprime, 2023-03-20:
Ola, tenho um mks dlc 32 e gostaria de mudar a logo inicial da firmware.

Medina-ui, 2023-03-20:
Boa tarde, não sou da MKS mas veja este vídeo que explica como fazer:https://youtu.be/GtvW54q12wcEnviado do meu iPhone

GabrielxSilva, 2023-05-03:
Pode fazer para mim também ? Uso TS35-R . Obrigado

[image]

K4o0s2, 2023-06-22:
> > 
> 
> Hay una forma muy sencilla y que cada uno de nosotros puede cambiar de forma muy sencilla. Próximamente insertaré una guía para reemplazar la pantalla de inicio personalizada.

I look forward to that method. The one that I have seen and tried is to compile the firmware with visualcode and customize the screen (there is a link to the video above) however, the latest source code does not correspond to the latest binary released and that is a problem for my. I recently proposed to makerbase the option that the update process itself search for the microsd image file and if it finds it, it installs it and if it doesn't put the standard but they haven't answered me. Well yes, they have given me the image that they do to me (they have never said when) 😅🤣🤣

sharknetwork24, 2023-06-22:
Fully functional and definitive guide.

https://github.com/makerbase-mks/MKS-DLC32/issues/390

sharknetwork24, 2023-06-22:
> I look forward to that method. The one that I have seen and tried is to compile the firmware with visualcode and customize the screen (there is a link to the video above) however, the latest source [...]

😂😂😂😂😂

sharknetwork24, 2023-06-22:
I saw that video and it's not simple to compile with VScode.
Not everyone understands unfortunately.
That's why I wanted to find a feasible and simple method at the same time.
I did some trial and error initially and the images weren't seen by the board.
Then I realized where I was wrong and I made this guide for everyone.

K4o0s2, 2023-06-22:
I'll try it in a few minutes and tell you. Thanks a million for the work done.
Greetings from Spain

suganth1999, 2025-03-16:
I need to custom my TS35 boot screen for mks dlc32 v2 CNC firmware