makerbase-mks/MKS-DLC32 #128: Running mobile app with DLC32

daven78, 2022-02-09:
Hello,
Some of the buttons in the mobile app don't display correctly on my iphone.  The formatting isn't correct.  Is this an issue you are aware of and correcting?  See attached picture
[image]
.

abinoer, 2022-02-10:
@daven78 What kind of iPhone do you have and what is the system version numbe?

serial-tintypist, 2022-02-10:
> @daven78 What kind of iPhone do you have and what is the system version numbe?

It looks exactly the same on my iPhone X, iOS version 14.7.1)

daven78, 2022-02-10:
Iphone 11

Software version 15.1

daven78, 2022-02-11:
i fixed it by changing the language to Chinese then back to english and it fixed it.  Must be a bug in the language conversion routine.

abinoer, 2022-02-11:
@daven78 Thanks for your feedback and we will fix it

closed 2022-02-11 by daven78