makerbase-mks/MKS-DLC32 #129: German translation Missing Text and wrong Translations

Lasermax3000, 2022-02-16:
Hello MKS-Team, 

the German translation on the Touchscreen needs more improvements to be lets say ...usable.
There are many words missing, where it says only Text. If you need some support i would like to help you improving the translation.

Lasermax3000, 2022-02-19:
No the English Version is correct. But you added in on of the last Versions the option to switch in Config File to German language, if´ve tested this and at the moment it is not working pretty well. So i made an new Issue. I´ve just wanted to help you making a better translation. According to the new Issue #133 i will upload you the translatet .txt File if this is helpful to you.

abinoer, 2022-02-21:
Oh! Sorry, there must be a configuration file mistake, there is no German at present. If convenient, I hope you can help to translate

closed 2022-02-21 by abinoer