makerbase-mks/MKS-DLC32 #132: Editing configuration.h file

kdaffy, 2022-02-17:
Hi, is it possible to edit the configuration.h file and where could i find it? The firmware files are ''bin'' files. Does anyone know how can i do this? In my 3d printer i easilly do it with ''arduino ide'' or ''visual studio code''. Here how do we edit it????Thanks!!!!

makerbase-mks, 2022-02-19:
Can this help you?https://github.com/makerbase-mks/MKS-DLC32/wiki/Firmware#configure-through-configuration-files

kdaffy, 2022-02-19:
Thanks but i already know this. This is the basic configuration. I need the configuration.h file which has all the settings as for example what stepper drivers you use or the configuration of limit switches and a many many more as we do in a 3d printer. Thanks!!!
See this example :  https://marlinfw.org/docs/configuration/configuration.html

torenich, 2022-02-19:
@kdaffy 

These boards do not run Marlin. They run a version of Grbl for ESP32. A full list of settings can be found here...
https://github.com/bdring/Grbl_Esp32/wiki/Settings

Pretty much any setting you need to change can be done with extended settings form the console.

This is currently the only was to change settings until MKS releases the source.

kdaffy, 2022-02-20:
Thank you very much. This is what i wanted!!!!!

closed 2022-02-20 by kdaffy

josmeijer, 2024-07-22:
What I miss here is the configuring through the web interface 
[image]

josmeijer, 2024-07-22:
[image]