makerbase-mks/MKS-DLC32 #133: Ask users to help translate

abinoer, 2022-02-19:
[language_en.txt](https://github.com/makerbase-mks/MKS-DLC32/files/8101636/language_en.txt)Ask users to translate it into whatever language you need, German, Portuguese, Spanish, etc., and upload it here. Thank you very much

manuelchaves, 2022-02-19:
Hi, I can do the portuguese translation, no problem. Just give me the files and instructions.

manuelchaves, 2022-02-19:
Finished translation = Portuguese -BR, but I don't know how to upload the .TXT file, can you help me?

manuelchaves, 2022-02-19:
Finished translation = Portuguese -BR, but I don't know how to upload the .TXT file, can you help me?
@abinoer

abinoer, 2022-02-19:
Click on "Attach Files by dragging & Dropping, selecting or pasting them." at the bottom of the comment

Lasermax3000, 2022-02-19:
[language_de.txt](https://github.com/makerbase-mks/MKS-DLC32/files/8102333/language_de.txt)
I made a German translation. What is meant by the Roller Mode? I was not sure about what that means?
And another question, what is the difference in the Control menu between the Postion and Home Button?

manuelchaves, 2022-02-20:
Portugues,Br version
@abinoer 
[language_PT-BR.txt](https://github.com/makerbase-mks/MKS-DLC32/files/8103976/language_PT-BR.txt)

torenich, 2022-02-21:
> Roller Mode

Probably means a rotory axis. Like for engraving round objects. That would be my guess.

johnrodrit, 2022-03-03:
Spanish translation anne
[language_ES.txt](https://github.com/makerbase-mks/MKS-DLC32/files/8180805/language_ES.txt)
x

cipyyy, 2022-12-18:
Romanian translation with diacritics
[language_ro.txt](https://github.com/makerbase-mks/MKS-DLC32/files/10252947/language_ro.txt)

Romanian translation without diacritics
[language_ro 2.txt](https://github.com/makerbase-mks/MKS-DLC32/files/10252951/language_ro.2.txt)

closed 2023-01-09 by abinoer

azwing, 2023-02-19:
Hi may be this can help 
French translation.

[language_fr.txt](https://github.com/makerbase-mks/MKS-DLC32/files/10776500/language_fr.txt)

tasiorg, 2023-12-20:
[language_pl.txt](https://github.com/makerbase-mks/MKS-DLC32/files/13731670/language_pl.txt)
Polish translation :)