makerbase-mks/MKS-DLC32 #134: Adding Enclosure Switch Open/Closed

Lasermax3000, 2022-02-19:
Hello MKS-Team,

this shall not be a new Issue, even more a enhancement in Safety. 
I would like to know if there is any free IO-Port (may be SDA PIN) on the DLC32 to connect a switch for detecting if a Laser or CNC-Machine Enclosure is open or closed. So that you can push the Start Engraving Button and the Engraver waits until the Enclosure is closed before starting the engraving process. And Additionally when you open the enclosure it interrupts the e.g. Engraving till the Enclosure gets closed, as a Safety Feature.
Please let me know if you got any Ideas, how the realise this function.

abinoer, 2022-02-21:
All IO ports are already in use, there is no extra IO to implement this

Lasermax3000, 2022-02-21:
Sad to hear, but why can´t the SDA Pin be used? I think it actually is only used to get in Flashing Mode at Startup. 
Have you another idea how to realise this function? Maybe with the Reset Button or Probe Switch?
Thank you for your support.

abinoer, 2022-03-29:
No, we can't. It'll interfere with other functions. I'm sorry

closed 2022-03-29 by abinoer