makerbase-mks/MKS-DLC32 #135: Laser pulse when moving the X Y axes

Devil-67, 2022-02-19:
Hello all,

my new laser is almost ready for use.
However, I have noticed a problem when the X or Y axis is moved without the laser being switched on, there is always a very short and also very weak laser pulse.
This is dangerous and should not be the case. Does anyone have an idea what could be the reason for this?
I use two Mean Well power supplies, one for the controller and one for the Neje laser head.
The rack is earthed.
I suspect it is coming from the MKS board but I am not sure.

Lasermax3000, 2022-02-19:
Is the Controller turned on or also off? When the stepper motors are moved the induce voltage and when you have two seperate Power Supplies and they are not well grounded, your GND potential on the Board varies and that might cause this Problem. 
Are the GND Outputs of the two Power Supplies connected together or a both grounded, if not try this?
I hope that this helps you finding your Problem.

Devil-67, 2022-02-20:
Axis control and laser controller are switched on.
GND of the two power supplies are not connected to each other.
Both are earthed
Mean to connect GND of both power supplies

Lasermax3000, 2022-02-20:
Yes, i would reccomend you to try this.

Devil-67, 2022-02-20:
Connecting GND from both power supplies did not solve the problem.
Nothing has changed.
Even if I remove GND from the TTL cable which is used to control the laser, it does not change - the laser switches on again and again briefly with very low power when the axes are moved.
 
Firmware Version
[image]

TTL and GND connection from the MKS board to the Neje Laser Control Board.
Power line + not connected, because an extra power supply is used as recommended here.
[image]

Neje Laser Control Board with power supply connection 1
Power supply unit 2 connection to the MKS 32 board
[image]

Maybe I have connected it the wrong way ?
However, the laser works fine in cutting and engraving so I think it will be connected to the right post?

Devil-67, 2022-02-20:
Here is a video to help you understand what I mean and where the problem is.
https://user-images.githubusercontent.com/31169133/154863149-84117979-551b-4fa9-874a-8911eb733d2c.mp4

Devil-67, 2022-02-20:
Now I have just measured the voltage at the TTL signal at S1000 and it comes out 8.6V shouldn't it be 12V ?

Can I also use the spindle power connection as an alternative, this supplies 12V at S1000?

Devil-67, 2022-02-20:
The instructions from the Neje say:
[image]

abinoer, 2022-02-21:
@Devil-67 Are the GND of MKS motherboard and GND of laser board connected properly?
[image]
Try connecting the picture link to see if it works

abinoer, 2022-02-21:
> Now I have just measured the voltage at the TTL signal at S1000 and it comes out 8.6V shouldn't it be 12V ?
> 
> Can I also use the spindle power connection as an alternative, this supplies 12V at S1000?

I don't quite understand what you're saying.  TTL voltage can be up to 5V, why 8.6 V

Devil-67, 2022-02-21:
> @Devil-67 Are the GND of MKS motherboard and GND of laser board connected properly? [image] Try connecting the picture link to see if it works

Yes GND is connected at both points, I just checked again.

Devil-67, 2022-02-21:
So far I have not been able to find anything to change the problem, it is always the same, when the X / Y axes are moved the laser keeps triggering a weak laser beam at irregular intervals as shown in the video above.

Here I have looked at the TTL output on an OSZI.

In the blue areas the axes are stationary, the red area shows when the axes are moving.

[image]

torenich, 2022-02-21:
Is the wiring that goes to the laser head bundled together with the stepper wiring runs? It may be inducing voltage into the TTL wire. Those NEJE lasers will work with 3-12v for the TTL input so maybe separate the wiring going to the laser and re-test.

Devil-67, 2022-02-21:
the MKS and the Neje have their own power supply, GND is necessarily connected to the MKS board for TTL.
How do you mean disconnect 
[image]

Devil-67, 2022-02-21:
> Ist die Verkabelung, die zum Laserkopf führt, zusammen mit der Schrittmacherverdrahtung gebündelt? Es kann Spannung in den TTL-Draht induzieren. Diese NEJE-Laser arbeiten mit 3-12 V für den TTL-Eingang, also trennen Sie vielleicht die Verkabelung zum Laser und testen Sie erneut.

Now I understand, no the wiring is not together with the stepper motors.

abinoer, 2022-02-22:
[image]
@Devil-67 The voltage burr on your picture is only about 0.6V peak-to-peak value, far less than 5V, which should not meet the REQUIREMENTS of TTL signal. Why is there weak laser

mejson80, 2022-03-21:
I've had the same issue, you need to use shielded cable, 
TTL from DLC is 0-5V, I'm suspecting stepper wires induct current in the TTL wire, hence it's pulsing or going on low power beam, was the same when I touched the TTL wire - laser came on, anotherwords, use a shielded 4wire cable, connect shield with GND, that solved my problem, I've extended my signal cable from original one to 1.5m ;)