makerbase-mks/MKS-DLC32 #136: More Information on the MKS DLC32 v2

Hargard339, 2022-02-20:
Hi. Its the V2 board that I can find only very little information online regarding this board. 
Yes its quite a new board. 
What I have found is :- https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf
https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20Firmware%20Programming%20Instructions.pdf
https://github.com/makerbase-mks/MKS-DLC32/wiki/Firmware
Very little on setting up drivers.. IE TMC2209 and other drivers. 
There are No videos for the V2 version. 
Only this one which is Not the V2 version. https://www.youtube.com/watch?v=G1q-eSL_V_I
Thanks in Advance. 

Regards Hargard.

abinoer, 2022-02-21:
Do you mean the breakdown Settings for various drivers?

TheExpertNoob, 2022-02-22:
...and differences between the v2.0 and v2.1 board.

Hargard339, 2022-02-24:
Hi. Thanks. I now have my board and screen. Will use DRV8825 drivers so start with. 
With no method of setting up the drivers Not sure the board is all its cracked up to be.. 
Start tomorrow pm sometime and just connect everything up as shown in the video for the V1 board.

abinoer, 2022-02-25:
!
[image]
[1645777937(1)](https://user-images.githubusercontent.com/96047125/155682168-95f0bd3c-e906-4b11-b0a8-4400677d0955.png)
@Hargard339 All dip switches are upturned. DRV8825 is 1/32 step

abinoer, 2022-02-25:
@TheExpertNoob 2.1 version compared with 2.0 version, 2.1 has gold sinking process and more detailed screen printing, there is no difference in other places

VukIgor, 2022-03-26:
Hi ,got new board via aliexpres dlc32 v2.1 with ts24 and tmc2209 ,installed firmware,runnded lightburn and hit an issue ,"feedrate is not defined " and in lasergrbl something like "run grbl on another type of board" ... in lightburn every movement was in reverse even if it shouldnt be becouse before mks board i had ramps 1,4 running without issue with grbl . There isnt any documentation for that board and it is waste of time and $60 becouse thing dont work as it should with provided laser firmware from makerbase

Hargard339, 2022-04-02:
The board on first start was not interfacing to the screen very well. Inverted and upside down. This was fixed by flashing new firmware to the board.. Easy fix.
The installation of the TMC2209 drivers... Required no setup or messing around. Basically they are plug and play with this board other than the Microstep settings.
All good with the settings.   
The only thing I dislike about the board is the Power supply input that is at the front. as this spoils the enclosure panel having a plug and cable at the front of the machine. Needs one at the sides or rear of board.
A good product. But documentation could be a lot better.

Thanks... for a great product.
Hargard.

closed 2022-04-02 by Hargard339

Rickrat, 2023-04-17:
Harvard, there are holes on the MB so you can solder another pigtail jack and route it out the back.