makerbase-mks/MKS-DLC32 #139: DLC32 WIFI hangs when engraving : using Lightburn via VSP

3Dangel, 2022-02-23:
Dear all
reporting with my issue here:
SW: Lightburn
HW: K40 Co2 Laser
Ctrl Board: MKS DLC32 V2.0

issue:
engraving a job using Lightburn , via Virtual Serial port to connect via WIFI to DLC32. 

The DLC32 after starting the job about 10 sec. will stop and will restart.
Job cannot be completed.
All basic function of movements via WIFI without issue. 
Only when sending a job to engrave will stop the operation.
please any help?
thanks!

abinoer, 2022-02-25:
Our r&d people will look at this problem when they have time

AngleV, 2022-02-28:
I confirmed that too.
DLC32 v2.1 using LightBurn via Virtual Serial port to connect via WIFI to DLC32 its starts working and freezes after a while.

geogarden, 2022-05-22:
I confirmed that too.

johntruman, 2022-07-24:
I'm also having this issue.

I've tried various gcodes, both engraving and cutting, generated by Lightburn, Laserweg, and Lasergrbl.

Operation freezes at some point, and not always at the same place.

selam, 2022-08-28:
With V2.1 board and laser engraving firmware is V2.10_H35_2022_0621_N_ZX_001, also wifi connection is not working at all, wifi connection is ok but can't connect via lightburn or via web browser.

frobert, 2025-01-16:
Same issue here no mater what i do, works prefectly fine via direct USB... Any new info on that issue?

Highlander-UA, 2025-05-17:
I can confirm, problem is still here. Hangs at random moments trough WiFi, works fine over USB.  Some useful info can be gathered in a such way: connect DLC32 with Lighburn trough WiFi, and at the same time connect board with USB cable to computer. Connect to DLC32 USB serial port with ordinary serial terminal software (like PUTTY or Realterm, for example). And than start job in Lighburn. When crash occurs, You'll see crash dump info in serial terminal. In my case it's like this:

assertion "pbuf_free: p->ref > 0" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c", line 766, function: pbuf_free
abort() was called at PC 0x401292af on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008af5c:0x3ffd1cf0 0x4008b1d9:0x3ffd1d10 0x401292af:0x3ffd1d30 0x401673e3:0x3ffd1d60 0x4015ee4a:0x3ffd1d80 0x4015f92d:0x3ffd1e00 0x4015f9a2:0x3ffd1e30 0x400f29ee:0x3ffd1e50 0x400f2a59:0x3ffd1e70 0x400f2b11:0x3ffd1e90 0x400df6a8:0x3ffd1eb0 0x400e8da6:0x3ffd22e0 0x400e8c4c:0x3ffd2300 0x400daf47:0x3ffd2320 0x4008ce8a:0x3ffd2340

Rebooting...

PC address is the same, backtrace addresses differs from time to time.