makerbase-mks/MKS-DLC32 #142: Some Issues

lasermanx, 2022-02-25:
Hello,
I just setup my board for a DIY diode laser today. I am having some issues. 

The movement of motors is not smooth ie. jittery. 

Also the laser Test Fire in Lightburn turns off  the laser after 15-25 seconds. 

Anybody experiencing similar issues?

abinoer, 2022-02-25:
1.  Maybe the motor wiring sequence is wrong. Please check it.
2.  The laser will turn off after more than 10 seconds because of the protection function, which is normal.

lasermanx, 2022-02-25:
1. I have checked the wiring. Engraving is perfect but movement of motor is not very smooth.
2. I want extended time for focusing of the diode. How is this possible in lightburn?

lasermanx, 2022-02-25:
Tried lasergrbl M3 S100 and the laser turns off after a few seconds. How am I supposed to focus the laser?

abinoer, 2022-02-25:
@lasermanx 
1. What is the specific situation of motor unsmoothness you mentioned? Is there any video
2.  You can adjust the distance of the lens or laser head to the material, then turn on the laser to observe the focus, and adjust accordingly

lasermanx, 2022-02-25:
I am not able to upload the video here so I have sent it over Facebook. 

Is there any way to disable safety because I need to focus very delicately simply measuring distance does not help. 

I don’t want this safety feature. Can you tell me how to disable this. I want to manually turn off laser in Test Fire mode.

abinoer, 2022-03-03:
In fact, this will not affect the focus too much, we can only disable this protection function by changing the firmware, but our r&d personnel are very busy at present

lasermanx, 2022-03-07:
I am not able to set $33 and also M4 commands are not working. Do you think I have a bad board.

abinoer, 2022-03-09:
The $33 setting has already been deleted, probably not updated on his wiki. If you want to set the spindle pulse rate, go to $28. The M4 works, but it only seems to work while on the move.

lasermanx, 2022-03-09:
Thanks for the clarification. The Test Fire turns off sometimes after 3 seconds, 15 seconds, 18 seconds or 28 seconds. It is very random. This is normal or do I have something going wrong?

abinoer, 2022-03-10:
@lasermanx Use the M4? It only turns on when it's moving. It turns off when the motor stops. Therefore, the length of movement time determines the laser on time

closed 2022-03-17 by abinoer