makerbase-mks/MKS-DLC32 #146: Will there be open-source firmware?

maxholgasson, 2022-02-28:


giordy72, 2022-02-28:
+1 for code release

serial-tintypist, 2022-03-08:
+100

daxliniere, 2022-03-21:
Yes, we could also submit pull requests here.

simulz, 2022-04-06:
source code가 필요합니다.
다른 용도로 사용하기 위해 새로 만들고 싶습니다.
Pin map이라도 알려주세요.

daxliniere, 2022-04-06:
@simulz, I think the pin map is already released, but not the source code, so we are unable to add things like true emergency stop button (not just reset ESP32) or add extra buttons to the GUI.

simulz, 2022-04-07:
@daxliniere Could you tell me where the pin map is? I can't find it here.
If you have MKS DLC32 ESP32 pin map, I can program it.

daxliniere, 2022-04-07:
@simulz have a look here: https://github.com/makerbase-mks/MKS-DLC32/tree/main/hardware

Specifically https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20SCH.pdf

daxliniere, 2022-04-07:
At the bare minimum, the Buzzer pin could be repurposed for something more important like a true emergency stop button. (In case you are not aware, using the existing emergency stop function just PUSHES reset on the ESP-32.)

CLP58, 2022-04-13:
+10000

geogarden, 2022-05-22:
+1000. When ?

ubergeekseven, 2022-08-27:
I am trying to move to fluidnc since this project is just a stolen version of esp32grbl. Trying to figure out pin mapping still though.

ubergeekseven, 2022-08-27:
What is the ttl pin gpio?

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks