makerbase-mks/MKS-DLC32 #148: Perplexity at laser commissioning, dimensions grow larger

Devil-67, 2022-03-01:
Hello all,

I am at a bit of a loss...
Today I put my new laser (self-built) into operation, control MKS DLC 32, software Lightburn, travel 1000X500 Two steppers on Y axis, so one per side.

At first everything looked good, then I cut 20X20 squares to calibrate the axes until the dimension was right.
BUT... when I had cut a height adjuster with 19, 20, 21, 22, 23, 24, 25, 26, 27, 30 in the form of a staircase, I noticed that all the dimensions deviated except for 20mm, to which I had calibrated. The 19mm dimension was undercut, and for all the others it was exceeded by approx. 0.01 per millimetre, increasing further and further, and for 80X80mm I was already at 80.50mm.
I have never seen anything like this before and I can't imagine where it comes from, let alone how I can fix it.
I looked in the control system but found no parameter that would indicate this, and I couldn't find anything in Lightburn either.

The belts are tensioned, the drivers are 4988 and I have already changed the voltage between 0.6 and 1.0V but the result is always the same and it affects both axes X and Y the same.
Power supplies are one 12V 12A Mean Well for the controller and the steppers and one 12V 5A Mean Well for the laser, voltage is stable power is abundant.
I had set the steps to 32 and then changed them to 16 for testing, same result, only I had to recalibrate, but it is also clear so far.

It would be nice if someone could help me with this.

Greetings

Devil-67, 2022-03-02:
Pulse settings ? where do I find them I only know microsteps

abinoer, 2022-03-03:
[image]
$100, $101 and ￥102

abinoer, 2022-03-03:
[](https://user-images.githubusercontent.com/96047125/156514688-2b7592f1-7c79-4e1a-812e-1106c8c4b66b.jpg)1646291451(1)
$100, $101 and $102

abinoer, 2022-03-03:
$100, $101 and $10
[image]

closed 2022-03-16 by abinoer