makerbase-mks/MKS-DLC32 #149: tcm2209 issue

kdaffy, 2022-03-02:
Hi, yesterday i received my tcm2209 from makerbase shop. I changed my old a4988 and changed the extended settingd $160, $161 and $162 from 16 to 256 (microsteps). They are very quiet but during an engraving suddenly it stopped engraving and i couldn't control the laser so i turned it off and on again. Does someone knows if when i change drivers i have to make any other settings? Thanks!!!

torenich, 2022-03-03:
That isn't going to work. I believe those microstep settings ($160-$162) are for UART or SPI drivers which the DLC32 doesn't support. If you have the standalone 2209s the highest you can set them for is 64 microsteps (according to the tables you referenced, but I've only seen them set to a max of 16) via the dip switch. If you set them for 16, the drivers interpolate this into 256 steps internally.

abinoer, 2022-03-05:
Yes,set them for 16, the drivers interpolate this into 256 steps internally.

Devil-67, 2022-03-06:
How is switch M2 set on the 2209 ?
It is not indicated in the table
[image]
.

abinoer, 2022-03-07:
@Devil-67 M2 is not indicated, indicating that M2 is ignored, and M2 high and low low are the same  @Devil-67

closed 2022-03-08 by abinoer