makerbase-mks/MKS-DLC32 #151: Laser and CNC on the same machine

serper, 2022-03-05:
https://github.com/makerbase-mks/MKS-DLC32/issues/119

A month waiting for a response to finally receive the answer "Sorry, only firmware can be replaced in this case" and close the question.

The idea is to be able to use the command M6 Tn (n = tool) to select the tool. You can use the I2C pins (two pins) and use relays, triacs, mosfets, ... according to the needs of each one.

In this firmware we have very big restrictions. All to be able to use the display, but to the answer "Sorry, only firmware can be replaced in this case" the reply is clear: OK, I change the firmware to FluidNC which does allow this and much more. I will lose the display but it is preferable.

abinoer, 2022-03-07:
Engineers thought the combination of laser and CNC would cause some problems, such as burning MOSFET, and after careful evaluation decided to continue to separate the two firmware. Sorry for the late reply

serper, 2022-03-07:
In my machine (for example) I have the laser in the connection for laser and the spindle with a relay connected to the SCL pin, the power here is external to the board. The idea is to be able to select the tool being able to configure in some way where each thing is connected. In FluidNC for example you can configure the GPIO4 as a relay where you will connect the spindle. Then with the M6 Tn command you select the tool, this allows you to combine the tool in the same job.

abinoer, 2022-03-09:
All GPIO has been used up and SCL has been used for M7/M8

closed 2022-03-21 by abinoer