makerbase-mks/MKS-DLC32 #152: Z PROBE PROBLEM

PEACE-CNC, 2022-03-06:
Hi, when Z Probe is used in programs, it crashes. Z probe latest version is not available. I have sent the relevant pictures
[image]
[image]
[image]
[image]
.

abinoer, 2022-03-09:
Which firmware are you using? Can the probe function on candle be used normally

abinoer, 2022-03-15:
Set $6 to 1, this function can be restored to normal

closed 2022-04-06 by abinoer