makerbase-mks/MKS-DLC32 #158: DLC32 with TMC2209 speed problem

Fusselbirne, 2022-03-13:
My new Controller DLC32 v.2.1 is coming with TMC2209 v2.0 driver. My small CNC 3018 with nema17 Stepper run not faster than 1400 ( gcode: G1 X100 F1400). With my old driver A4988 stepper is working up to 3000 speed. I try with 1/8 and 1/16 microsteps and vref from 0.8 up to 1.4v but nothing is working. Is V2 of this driver not compatible with that controller? What can I do to work with that driver?

abinoer, 2022-03-14:
Did you actually move the correct distance when you used 2209? Or maybe the maximum speed and acceleration are set differently,

Fusselbirne, 2022-03-14:
Yes I adjusts all grbl settings. When I move „G1 X100 F1000“ it’s moving 100mm, right. But faster than 1400 to move it’s not possible. Acceleration is 200. And with drivers A4988 it’s working faster. I have controlled the real speed with stop watch.

Ondrashek, 2022-03-14:
$100-$112 to 6000
$110-$122 to 500

Fusselbirne, 2022-03-15:
Yes it was presets, but in this case move over „G0“ all motors blocked.

abinoer, 2022-03-15:
I can't reproduce this problem. What operations did you do and then found this problem? Can you provide the video

Fusselbirne, 2022-03-15:
Sorry i can't do video, i'm not home until friday. 
But again from beginning. I have CNC 3018 from china with cheap grbl controller and cheap drivers (A4988) but it works fine. Then i order better controller, dlc32, with better drivers, TMC2209, for more power, quieter and more features, it was the plan. My new toy was coming and i load cnc firmware, connect all cables and check the grbl settings. It was different but plausible, ok let's try. I write " G1 X100 F500" in Candle console and juhu my spindle is on the way. But it was really slow and travel distance not right. Ok i have to adjust the grbl parameters, no problem. Set the $100 -$102 (steps/mm) from 80 to 400 like my old settings and try again. Ok it was better but not perfect. Speed was better but travel distance was half, not 100mm only 50mm. Ok i use 1/16 microstepping it's different. Give $100=800 and try again. Now is right. Travel distance 100mm speed (f500) like the same my previous controller and more quite, wonderful. Ok now we check full speed. Write "G0 X0" and.. my CNC is crying not moving. $110 -$112 (max mm/min) was 6000 preset, ok maybe to much for my litle CNC. I change $110-$112 to 2000 and ... right it's not moving. What??? wtf is it, it's worse like my old drivers. Cross check with my A4988 on DLC32 board. And right, only the direction is wrong but its going. More then 3200 (G1 X-100 F3200) it lose same steps. But 3000 without load it's save. Then i compare A4988 vs. TMC over stopwatch with speed 1000 (G1 X100 F1000). It is the same. I check the Y-axis and it's the same sch.... The TMC can't faster then 1400. And i have no idea why.

abinoer, 2022-03-16:
I am very sorry that I have not reappeared this problem until now,My Settings are the same as yours, Max speed set to 2000, acceleration 200, and 1/16 microstepping. Can the laser be turned on when the motor is not moving? Have you tried switching to another 2209? What if 2209 was damaged

Fusselbirne, 2022-03-16:
No I  have only one set A4988 drivers and one set TMC2209. But I tried TMC drivers on my old controller board and it was same result. The new TMC2209 is “grandpa version”.

abinoer, 2022-03-25:
We tried it here, 2209 does have a lower speed limit than 4988, and 2009 does have a weaker speed

Fusselbirne, 2022-03-25:
That’s not good news, anyway. And the newer type TMC2225 what have this for speed limit. Can I move faster with type, like 4988 or is that “grandpa version” to?

abinoer, 2022-03-28:
Sorry, this is the deficiency of 2209 chip, we can not solve it. If you want to use the 2209, you have to set the speed lower.

closed 2022-03-28 by abinoer

manitoulin-maker, 2022-08-03:
what is the speed rating for the tmc2225 and can the tmc2226 be used as well? its basically just a better 2209 right?