makerbase-mks/MKS-DLC32 #162: number of copies on TS35 not working properly

serial-tintypist, 2022-03-16:
MKS DLC32 v2.1, normal laser mode with a TS35 screen. When I upload a gcode file to the SD card and then run it from the TS35 screen, I can set the number of copies of the job. However, there is no pause between the individual runs, the laser finishes one copy and then runs another copy right away. There is no way to switch workpieces between copies this way, it works as multiple passes on the same workpiece instead (which makes no sense as you set the number of passes in software that creates the gcode in advance). Please add functionality that will pause the laser and wait for user confirmation between copies.

abinoer, 2022-03-21:
This is the same logic as software GRB to set the number of engraving.

abinoer, 2022-03-22:
[image]
Sorry, not GRB, but GRBL，LaserGRBL

closed 2022-03-28 by abinoer