makerbase-mks/MKS-DLC32 #164: CNC Firmware Z-Homing doesn't work

christian-hellwig, 2022-03-17:
When I issue a $H command only x and y home. Manual control of z-axis and hard endstops work. Could you please add the option for homing z.

christian-hellwig, 2022-03-17:
Hm that worked kind of. Now the z-endstop doesn't trigger

abinoer, 2022-03-18:
Sorry, we forgot to specify this.
Turn on:  $Homing/Cycle1=Z
Turn off:  $Homing/Cycle1=0

christian-hellwig, 2022-03-18:
Yeah the homing starts, but as mentioned the endstop doesn't work. Could it be, that the pin isn't assigned in the config?

christian-hellwig, 2022-03-18:
Okay that seemed to be an error with an older firmware. Updated to the latest CNC Firmware and now the endstop is recognized as well. Thanks for the support

abinoer, 2022-03-18:
Yes, the latest firmware I tried is ok

closed 2022-03-18 by abinoer

daxliniere, 2022-04-09:
> Can you try this command in lasergrbl? 
>$Homing/Cycle0=XY
>$Homing/Cycle1=Z
> This mean that Homing cycle will be XY axis together and than Z axis

Thanks @Ondrashek.

The safer option would be to home the Z-axis first.

$Homing/Cycle0=Z
$Homing/Cycle1=XY

Ondrashek, 2022-04-09:
> > Can you try this command in lasergrbl?
> > $Homing/Cycle0=XY
> > $Homing/Cycle1=Z
> > This mean that Homing cycle will be XY axis together and than Z axis
> 
> Thanks @Ondrashek.
> 
> The safer option would be to home the Z-axis first.
> 
> $Homing/Cycle0=Z $Homing/Cycle1=XY

For CNC sure. I am using XY first for laser :-)

adilrs, 2022-12-10:
> > Can you try this command in lasergrbl?
> > $Homing/Cycle0=XY
> > $Homing/Cycle1=Z
> > This mean that Homing cycle will be XY axis together and than Z axis
> 
> Thanks @Ondrashek.
> 
> The safer option would be to home the Z-axis first.
> 
> $Homing/Cycle0=Z $Homing/Cycle1=XY

Salvou minha vida ! :-D   Estava tentando a 2 dias

Hargard339, 2023-03-14:
Yea that works. Thank you.