makerbase-mks/MKS-DLC32 #166: Improvement for Laser GUI

Lasermax3000, 2022-03-18:
As an Idea i would like to add a new Button in the Adjustment menu, to turn Cooling or Air Assist on. And in addition to this point, could you please remove the 50% Laserpower on Button. In my opinion it is useless and dangerous for Adjustment to burn a hole in your Material before starting an engraving or cutting Job ;). I would rather reduce the Adjustment Laserpower to only 1%.

Ondrashek, 2022-03-19:
Agree. I need 1% laser power to TFT35 too.

The best solution will be some GUI for editing screen.

abinoer, 2022-03-29:
The cooling function will be added in the next version, and the laser interface power adjustment part will make more detailed adjustment methods.

Plottmania, 2022-04-01:
It would have something if M7 would work : )
A button to activate the assist which I currently still do directly from LightBurn via the activation in the layer to be lasered.

https://youtu.be/of7n8h_A5dU

pedropvbh, 2022-06-26:
bad only have 50% and 100%, you need to have 5% for positioning