makerbase-mks/MKS-DLC32 #167: Missing Static IP Settings

Lasermax3000, 2022-03-20:
Is it possible to set a static IP Adress to the Controller? 
If not could you please add this option to the config file in the next Version.

abinoer, 2022-03-21:
You mean arbitrary IP addresses?

abinoer, 2022-04-27:
The IP address is assigned by the router and has nothing to do with the firmware or motherboard. You can solve the problem as commented by the friend above, there is no other way.  sorry

closed 2022-04-27 by abinoer

Lasermax3000, 2022-04-27:
I´m sorry but that´s not true. Yes the IP Adress is assigned by the Router, but if a device wants a Fix IP it can request it.
My Router at Home is unfortunately not capable of assigning Fixed IP Adresses to Devices in it´s Settings. 
On other MKS Products such as the 3D Print Boards it is implemented in the Firmware, so yes it has something to do with the Firmware.
Here is an example how it´s done: https://randomnerdtutorials.com/esp32-static-fixed-ip-address-arduino-ide/
If you would make the Firmware OpenSource i would do it myself.