makerbase-mks/MKS-DLC32 #169: FR: Add step size button to Spindle screen

daxliniere, 2022-03-21:
On the Control screen, there is a button in the top left corner to toggle step size; 0.1, 1 & 10 mm.
This button should also be added to the Spindle page as I often have to move between these two pages to change Z-axis step size.

Also, it would be even more useful if the XY axes and Z axis had different step sizes. For XY, the button should toggle between 0.1, 1, 10 & 50mm. The Z-axis should toggle between 0.01, 0.1, 1 & 10mm.

Thank you for your consideration.

Best regards,
Dax Liniere.

abinoer, 2022-03-29:
I'm very sorry. After review and consideration of various factors, it remains unchanged

closed 2022-03-29 by abinoer

daxliniere, 2022-03-31:
If you share that part of the code, I will add the button myself and would be happy to send it back to you.