makerbase-mks/MKS-DLC32 #170: Hard limits and status reporting

daxliniere, 2022-03-21:
1) Visual feedback - At least on TS24 (I have no experience with DLC32 using TS35), the Hard Limit warning is too persistant. Once a hard limit warning has been dismissed, it is displayed on the screen again a moment later. This makes it difficult to jog the machine away from the limit switch because the dialog box keeps popping up on screen, covering the control buttons. The ideal UX is that once the warning is dismissed by the user, it should not be displayed again until a hard limit is triggered again.

2) Operation - I am using Universal Gcode Sender as the controller for my DLC32 in this situation. When a hard limit is reached, UGS receives a LOT of warning messages that prevent it from being able to talk to the DLC32 to control it. It kind of 'freaks out', preventing the user from jogging away from the limit switch.

As a UI/UX designer, I would be happy to consult on these and other matters.

Best regards,
Dax Liniere.

daxliniere, 2022-03-21:
Here's an example of the console when this happens. It only stops if I manually move away from the limit switch.

**** Connected to COM6 @ 115200 baud ****
Grbl 1.1 ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 7    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 1    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 11    (Homing direction invert, mask)
$24 = 20.000    (Homing locate feed rate, mm/min)
$25 = 100.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 0.500    (Homing switch pull-off distance, millimeters)
$30 = 24000.000    (Maximum spindle speed, RPM)
$31 = 8000.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$38 = 0   
$37 = 1   
$40 = 1   
$100 = 802.341    (X-axis travel resolution, step/mm)
$101 = 802.322    (Y-axis travel resolution, step/mm)
$102 = 802.497    (Z-axis travel resolution, step/mm)
$103 = 100.000   
$104 = 100.000   
$105 = 100.000   
$110 = 2000.000    (X-axis maximum rate, mm/min)
$111 = 2000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 80.000    (X-axis acceleration, mm/sec^2)
$121 = 80.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 305.000    (X-axis maximum travel, millimeters)
$131 = 190.000    (Y-axis maximum travel, millimeters)
$132 = 46.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $J=G21G91X100Y100F1000
ok
>>> $J=G21G91X-100Y-100F1000
ok
>>> $J=G21G91X-100Y-100F1000
ok
Grbl 1.1 ['$' for help]
>>> $$
[Error] An error was detected while sending '$$': (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended. Streaming has been paused.
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
[... 1446 lines cut ...]
$103 = 100.000   
$104 = 100.000   
$105 = 100.000   
$110 = 2000.000    (X-axis maximum rate, mm/min)
$111 = 2000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 80.000    (X-axis acceleration, mm/sec^2)
$121 = 80.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 305.000    (X-axis maximum travel, millimeters)
$131 = 190.000    (Y-axis maximum travel, millimeters)
$132 = 46.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Caution: Unlocked]

SlaUA, 2023-09-18:
Hi, did you solve the problem?