makerbase-mks/MKS-DLC32 #171: MKS DLC 32 Limit Switch connection

idilbaz, 2022-03-21:
Hello, I will connect to my mks dlc 32 board with two limit switches on each axis. Is it okay if I make the connection as in the following figure?

[image]

serial-tintypist, 2022-03-21:
Unfortunately, these particular switches will not work when connected in parallel like this (I know this from experience with my setup). You need to disconnect the red (5 V) line and use signal and GND only.

idilbaz, 2022-03-22:
In which way do I need to connect these shapes (A or B)
[image]
[image]

idilbaz, 2022-03-22:
> Unfortunately, these particular switches will not work when connected in parallel like this (I know this from experience with my setup). You need to disconnect the red (5 V) line and use signal and GND only.

In which way do I need to connect these shapes (A or B)

[image]
[image]

serial-tintypist, 2022-03-22:
B (green is signal and black is GND)

idilbaz, 2022-03-22:
> B (green is signal and black is GND)

Thank you so much :)

kweilim98, 2022-03-22:
I try already  with as u say but when I touch the limit switch, why it delay to respond? And after that my touch screen panel restart again problem happen, how?

idilbaz, 2022-03-25:
Hello, as you said, I made the connections as in figure B, but it does not work properly.  The Z axis limit switch does not work at all, and when the others touch the limit switch button, it gives an alarm in the UGS program, but it resets itself and starts the program again.  Why does the problem occur?  Did I make the connections wrong?

idilbaz, 2022-03-26:
My problem is solved.  The problem is with these limit switches.  They are not suitable for CNC machine.  I just used a limit switch, everything seems fine now.

kweilim98, 2022-03-29:
> My problem is solved. The problem is with these limit switches. They are not suitable for CNC machine. I just used a limit switch, everything seems fine now.

[image]

Bro, does u mean this limit switch module not suitable for the CNC?

idilbaz, 2022-03-29:
> 

Yes my friend.  I used Arduino CNC Shield before, and these limit switches did not work.  These switches did not work on the MKS DLC 32 board either.

closed 2022-05-16 by abinoer

mikaildogru, 2023-02-18:
hey friend how did you solve this probem?. I will plan door switch to connect sEmergency switches with series type ?. What do you think

kweilim98, 2023-02-19:
> hey friend how did you solve this probem?. I will plan door switch to connect sEmergency switches with series type ?. What do you think

What kind switch you using?

levent362727, 2023-12-27:
Merhaba ben mks dlc32 v2.1 kartim icin end stop kullanıyorum cihaz yeniden başlıyor neden oluyor bide gec algılıyor neden oluyor fikiri olan varmı