makerbase-mks/MKS-DLC32 #172: Request: Change Emergency Stop button behaviour

daxliniere, 2022-03-21:
Currently, the Emergency Stop button completely resets the ESP32. This results in loss of position and other data.

It would be best if this was reconfigured to pause the program.

Better still, since I'm using a locking 'mushroom-type' switch, it would be great if the program could be un-paused when the switch contacts were opened again.

[image]

kweilim98, 2022-03-22:
Agree with you suggest

abinoer, 2022-03-29:
I'm sorry, there is no extra IO to realize what you said

closed 2022-03-29 by abinoer

daxliniere, 2022-03-31:
Hello @abinoer, perhaps you have misunderstood my request. I'm not asking for an additional button to be added, but instead for the existing function to be re-written to pause the CNC, rather than resetting the microcontroller. Losing your positioning is a real pain, especially if you just wanted to pause the program.

As a side note, it seems that quite often it comes up that features can't be added because of a lack of available IO pins. It would be very advantageous to add 1) some extra input pins on the next version of this board and 2) support I2C-based I/O expander add-on boards.

Operater2803, 2024-12-01:
Seems old thread but Im currently serching for solution on this problem and i agree with daxliniere on this. My thinking is in way to use one of the unused IO ports like SDA pin it says that is used as cooling option port, maybe it can be programmed in firmware to be input port for E stop and just to halt operations. From my opinion this should exist for CNC firmware as not much use of cooling port here if one needs cooling just conect it together with spindle start relay. On the other side on laser firmware I used it as Air blow start during program run. Also on laser Firmware I think that probe IO22 pin is used as Fire detection and that is used to halt all operations so from my opinion can be used as E stop also. Correct me if im wrong.