makerbase-mks/MKS-DLC32 #174: homing asse z

massimop3632, 2022-03-24:
buona sera 
sto provando per la prima volta la dlc32 v2.1 su una fresa cnc.
ho impostato i parametri per homing ma l'asse z , si comporta come se fosse una macchina laser
ho caricato il firmware V2.0.8_H35_20220105_N_CNC
dove trovo il file config.h?

abinoer, 2022-03-29:
Do you mean to use CNC firmware and then Z-Homing doesn't work?Z-homing can be configured by sending commands
[image]

Gabri988, 2022-03-29:
@massimop3632 dalla console inserisci questa riga di comando: $Homing/Cycle1=Z e fai un homing

La macchina seguirà quest'ordine di azzeramento assi, XY e per ultimo Z. Non ho ben capito perché lo Z viene fatto per ultimo ma purtroppo è così 😅

closed 2022-04-01 by abinoer