makerbase-mks/MKS-DLC32 #175: G92 Z19.2 command problem

idilbaz, 2022-03-28:
Hello, I am using the UGS program. I made the Z probe connection to my MKS DLC 32 v2.0 card. The code g38.2 Z-10 F20 works. With this code, I can find the probe, but after finding the probe, the G92 Z19.2 command to go to the zero position does not work. Why might the problem occur?

abinoer, 2022-04-01:
[image]
Hello, my candle test is normal. What kind of host are you using

idilbaz, 2022-04-01:
> [image] Hello, my candle test is normal. What kind of host are you using

Thank you for your answer.  I detected the problem.  While the Z axis is going down, the tooth is skipping, so it misses the zero point.  Now I'm going to refresh the z-axis.

closed 2022-04-06 by abinoer