makerbase-mks/MKS-DLC32 #179: Limit pins invert issue

dhanninacs, 2022-04-04:
I have dual y axis stepper motors and two y limit switches on either side in Normally open position. So both switches need to trigger for hard limit to switch on. I'm using $5=1 to achieve this. 

But doing this, the x-limit switch which is also connected to Normally open and ground doesn't trigger when the limit switch is pushed. I have tried other limit switches to see if it is the problem but it looks like more of an $5 issues on board side. Did anyone have similar issues? I have a DLC 32 v2.1 board.

abinoer, 2022-04-06:
I don't quite understand your description. Are you in a situation where Y-limit works and X-Limit doesn't? Is only one Y Limit Switches and one X Limit Switches restored? Can you upload wiring pictures

dhanninacs, 2022-04-06:
Hi yes, when I have $5=1, and the X-axis limit switch connected to normally open position. The limit switch doesn't trigger when pushed.

abinoer, 2022-04-06:
Short X limit interface S and GMD, can trigger? If so, there's something wrong with your limit switch

dhanninacs, 2022-04-06:
I have tried other limit switches, they too don’t work. With $5=0 and limit switches connected in normally closed position everything works fine.

dhanninacs, 2022-04-07:
It was a wiring issue. All sorted out now and works as intended. Thanks

closed 2022-04-07 by dhanninacs