makerbase-mks/MKS-DLC32 #184: Open Source Firmware

CLP58, 2022-04-13:
You already said last October that you were ready to open source and it has been six months since then? When will you open source the firmware?
[image]

MTRatcliffe, 2022-05-08:
Would be nice to see this shared and to see what comes from it.

geogarden, 2022-05-26:
Promises must be kept :)

Hargard339, 2022-05-29:
Yes please To many issues for this Board. Make it open source.
It helps you and it helps Us.
Regards.

Lasermax3000, 2022-06-07:
That would be great!!!
Don´t know why MKS doesn´t want the help from the Community supporting there Board and making there Firmware better for free. 
I think if the Firmware would be better and you would have more options for customizations the Board would be more atractive for people out there -> potential buyers....just saying.

pedropvbh, 2022-06-26:
it's past time to release

Hargard339, 2022-06-26:
Wake up MKS we are here to help you as well. Listen to your supporters/customers.
Open Source is the way to go.

skaterdude, 2022-07-01:
Considering buying some of these boards with displays for various projects.  Without open source software, however, it's easier to continue with my own custom breadboard.

distebia, 2022-07-07:
ho bisogno di aumentare la mia uscita minima del pwm, su arduino impostavo spindle/pwm/min a 25 e risolvevo il problema...ora sono su mks dlc32 v2.1, la macchina è la stessa quindi con il solito problema dell'alimentatore co2 che vuole aumentata l'uscita minima, e mi sento dire che i parametri $30=1000, $31=0, $35=0 sono bloccati e, anche se vario i valori, non ottengo nessun cambiamento. un conto è avere un firmware protetto e non pubblico, un conto è avere variabili del firmware che non si possono modificare...non abbiamo tutti la stessa macchina laser o cnc, ogni utente è diverso e richiede parametri specifici.

dixi83, 2022-10-02:
> Why not making the source open merging this development work with FluidNC?

Why not see if we could get FluidNC working ;) waiting for MKS is a useless case. There is a config file provided ( https://github.com/bdring/FluidNC/blob/main/example_configs/MKS_DLC32_21_XYZ.yaml ) But could net get it to work on my V2.1 with TMC2209's. Because I am still building my machine, I did not spend many time on this. I needed movement so I switched back. I will give it another try when the machine is finished.

mikerocklewitz, 2022-10-05:
Im with everyone here.  Open the source.....or at least give us a good config for both cnc and laser, with some info for using different drivers and spindle/laser options with fluidnc.

Ive had this board sitting in my drawer for over 8 months now. I wanna use it on my 3018. Ive  tried getting the provided firmware to work but was met with nothing but issues.  Then I tried getting fluidnc to work but man thats a nightmare building a config off of the xyz laser example they have for this board. Nothin but issues.

Give us the goods already! Ull sell more

skaterdude, 2022-10-05:
I gave up.  It's not worth waiting, and I've got lots of other projects to do.

Hargard339, 2022-10-06:
I have as well. Threw my MKS DLC in the junk box as the firmware is not fit for purpose.   Will NOT purchase a MKS board again without first checking that we can adapt it to our requirements.  
WAKE UP MKS.

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks