makerbase-mks/MKS-DLC32 #185: pinout

guestisp, 2022-04-14:
i think some pin are missing in the drawing.
in example, which pins are used by en, step and dir for X and Z motore?

the tinybee has a table listing all pins used, but what about the dlc32? i have to create a custom firmware but i don't know the pin mapping

abinoer, 2022-04-15:
Hello, you can see all the pins by looking at the schematic

guestisp, 2022-04-15:
Are you referring to this ? https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20PIN.pdf

where are the pins for the motors ?

I can see, in example, the pins for I2C: IO0/IO4

But the pins for the stepper aren't shown. In example, which is the enable pin for X motor ?

abinoer, 2022-04-19:
https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20SCH.pdfThis might help you

closed 2022-04-29 by abinoer