makerbase-mks/MKS-DLC32 #186: Limit switches X + Y

bbuster2, 2022-04-17:
Which endstops can i use to detect the end of the X and Y axis?
And can i use it on both sides or is it just on one side of each axis to auto home?

I would like to limit the X,Y on the 0,0 and also X,Y

Can i also use inductive limit switches like https://www.benselectronics.nl/inductieve-proximity-sensor-npn.html?source=googlebase&gclid=CjwKCAjw9e6SBhB2EiwA5myr9pcqY3RnFE_lrFCwpA8g8KPvM5aR5HPPRZnzkir1G-NtMWY1XOa3choCLGUQAvD_BwE

Plottmania, 2022-04-18:
You can use this on a DLC32...no Problem : )

https://youtu.be/KuEqwmGYVGg

bbuster2, 2022-04-18:
Are you sure, the ones i posted are not the ones in the video, he uses PL-08N.
And am i also able to use 2 sensors per axis to limit to both ends of the axis?

Plottmania, 2022-04-19:
Why, I use there in theVideo inductive limit switches for a reference run.
I do not need limit switches at the end of the axes because I use the workspace monitoring.
If then you can connect limit switches per axis in series, which does not work with inductive or I have not yet tried.
Why also, is also enough so and needs no other switches.

bbuster2, 2022-04-19:
Thank you for the help, can you explain more what "workspace monitoring" you use?

My idea for multiple switches was to make sure it doesn't go too far in any direction. I was also thinking about putting a limit switch on both Y-axis

Plottmania, 2022-04-19:
In GRBL I have the option for SoftLimits (workspace monitoring).
I use LightBurn, drive a reference, have my zero point at "absolute coordinate" and from there a defined area.
If a workpiece is outside this range, I get a message.
So I don't need a limit switch at the ends of the axes.
One limit switch per axis is enough for the reference run....the rest, as I said, is done by the software.
I also have this option in DrufelCNC and other control programmes.

Sorry for my bad english : )

bbuster2, 2022-04-19:
Thank you and don't worry, your english is good.
I use Fusion360 so i am not sure how i could limit my workspace.

I will buy the limit switches (1 for each axis) and try if i can home and then limit my workspace with softlimits. Maybe the GRBL has bed dimensions built in from the home point?

Plottmania, 2022-04-19:
I haven't tried connecting limit switches in series yet, but it is possible.
Since I only use inductive ones on my DLC controllers: 5 V, GND to signal (range of inductive limit switches here 4 - 30 volts).
But you can connect normal limit switches in series with GND and signal...
If you use a DLC32 either FluidNC or GRBL and there the question which transmitter?
You can specify your working space in the GRBL without any problems.
I know Fusion but don't work with it...I use LightBurn.
But all only for lasering but I think a size can also be defined in Fusion or : )
You only create your file with it and it has to fit into the workspace, no matter which channel.
Anyway, if I have time these days I'll do a test setup for you to see if it works.

Plottmania, 2022-04-19:
Ein kleines Update von meiner SainSmart ProVerXL 4030...demontierte Endschalter und deren Verkabelung.
Kein Plan ob es so funktioniert mit einem DLC32 aber an der Fräsmaschine ohne Probleme mit dem Controller 0/8/15 und GRBL 1.1H

[image]

bbuster2, 2022-04-19:
Thank you for showing! 
I will see if i can implement it on my own device

closed 2022-06-02 by abinoer