makerbase-mks/MKS-DLC32 #187: X,Y, Z probe

kweilim98, 2022-04-18:
Does the features will update with 3axis probe? 
[image]

abinoer, 2022-04-20:
Do you mean the z-axis tool probe here

kweilim98, 2022-04-20:
No, I mean the X，y and z touch probe, to set the workpiece coordinate in origin and more precise

kweilim98, 2022-04-20:
Now only have the z tool probe only

daxliniere, 2022-04-23:
@kweilim98 Assuming your 3-axis probe has a sinlge output, it can all be done with a macro in UGS.

This is my Z-axis probe macro, but it could conceivably be modified for X and Y axes:

```
G21
G91 G38.2 Z-46 F100
G0 Z1
G38.2 Z-2 F50
G10 L20 P1 Z4.985
G10 L20 P2 Z4.985
G10 L20 P3 Z4.985
G10 L20 P4 Z4.985
G10 L20 P5 Z4.985
G10 L20 P6 Z4.985
G90 G0 Z20
G90
```

closed 2022-06-02 by abinoer