makerbase-mks/MKS-DLC32 #188: PWM to Analog for spindle

kweilim98, 2022-04-18:
Does the pwm signal at spindle port able to convert into analog 0-10v signal? Because my spindle need that to regulate the spindle speed

abinoer, 2022-04-19:
0-10V? This needs to buy a 5V PWM to 0-10V module, connected to the TTL signal interface

kweilim98, 2022-04-19:
Can u sent the pictures about the product?

abinoer, 2022-04-19:
Because all I found were pictures with Chinese captions, so you need to look for it yourself. Online should be a related module to sell, is about the PWM to analog module

kweilim98, 2022-04-19:
Is okay, I understand the Chinese, just sent me

kweilim98, 2022-04-19:
The the pwm signal need connect to the ttf port?

abinoer, 2022-04-19:
tt port.
[image]
Theoretically, this module can be used, but I have not used it, so you need to consult the seller and try by yourself

abinoer, 2022-04-19:
TTL por
[image]
Theoretically, this module can be used, but I have not used it, so you need to consult the seller and try by yourself

abinoer, 2022-04-19:
[image]
Theoretically, this module can be used, but I have not used it, so you need to consult the seller and try by yourself.TTL port

kweilim98, 2022-04-19:
Mean the pwm signal input from the TTL port rite?

abinoer, 2022-04-19:
Yes, the TTL port outputs a PWM signal with a peak value of 5V

kweilim98, 2022-04-19:
This one module can?

abinoer, 2022-04-21:
This I also have not had, dare not guarantee can, just provide an idea

abinoer, 2022-04-21:
This I also did not use, dare not guarantee can, just provide an idea

abinoer, 2022-04-21:
This I also have not used, dare not guarantee can, just provide an idea

Patsin1981, 2022-04-21:
Who have wiring diagram for TTL pin -> 5V PWM to 0-10V module -> Inverter?