makerbase-mks/MKS-DLC32 #190: Laser ttl output max power

ahmetbozaci, 2022-04-20:
- TTL output support max 25w/50w. Is this laser input power or output power? Because many powerful diode lasers have 30/40/50w input power. If it is so what can we do, should we use the relay module?

[image]

Plottmania, 2022-04-20:
Then try connecting 24 volts to a laser that has a driver and only likes 12 volts :)
Only very few modules like some Ortur run with 24 volts and 2 amps...because the driver can do this.
If I only connect 12 volts, the fan, for example, will no longer run properly.
But I think if you put 24 volts on a 12 volt board (driver) you won't have any fun with your laser!

Why, because it's too much voltage and the built-in fans don't like too much of it.
Apart from that, there is an output for lasers on the controller and if I do, I buy a laser that can switch 12 / 24 volts, such as the lasers from OptLasers.
In my opinion, however, they do not belong on the spindle and even if they did....wouldn't the laser run permanently there?
What do I do with $32=0 or 1?

manuelchaves, 2022-04-20:
hi, never use a laser, motor or any other high power (W) directly on the board or any other. The ideal is to use a relay (solid or conventional state). It's like 3D printers, you never connect the hots (hotend or hot bed) directly to the board. Go through a power module

See yhis
https://github.com/makerbase-mks/MKS-DLC32/issues/178

ahmetbozaci, 2022-04-20:
@manuelchaves Thank you for your answer. Can ı use mosfet as a relay module? 
[image]

abinoer, 2022-04-28:
[image]
@ahmetbozaci Hello, what kind of control does your laser head use

ahmetbozaci, 2022-04-28:
@abinoer 
For example, this is for neje laser modules. But some laser module has not this board. 
[image]

abinoer, 2022-04-28:
[image]
The TTL signal is connected to the corresponding pin of the DLC32, and the power supply is external. Do not obtain it from the DLC.

ahmetbozaci, 2022-04-28:
@abinoer Ok what if the laser has not this board. The laser has only 3 pins and it can connect to the dlc32 laser pin very well.
Also, another user said to use a relay, ı was thinking of using MOSFET.

abinoer, 2022-04-29:
Are the three pins TTL, GND and 12V respectively? That is also according to the connection I said above.

Plottmania, 2022-04-29:
Each DLC has a connection for lasers that should NOT be connected via the spindle!
3 pins for 12 V - GND - signal
All laser modules have their own driver and now you just have to check which supplier!
At Ortur there are modules that need 24 volts and max. 2 amps.
But again, a laser does not belong to the connection of the spindle!
The Neje board is an option for Neje laser modules, but you don't really need it!
What it needs and has every driver as an input: 12 volts - GND - signal...you just have to pay attention to the pin assignments.
Fits on every DLC....on every!

VukIgor, 2022-05-10:
I have Mks DLC 32 v2.1 and had issue with laser power ,laser blinking on G moves and my laser module was connected to mks motherboard . Solusion for that issue ,solution for 4wired laser module that is .
laser ( + ) wire conected to external power supply (  + ) terminal , laser signal ( ttl ) wire on S pin on 3pin laser connector on mainboard , laserer ( GND) wire goes to power supply ( - ) terminal  , laser temp wire terminate ,you dont need that .
To get full power of laser module take one wire ,connect one end to ( - ) terminal on psu and other to ( ttl ) pin on mother board where laser connector is ...
Enjoy full power without blinking or low power issues

closed 2022-05-16 by abinoer

Skipcruz, 2024-06-06:
@abinoer  foto .... 👀