makerbase-mks/MKS-DLC32 #196: Change type button in adjustment menu

Zener28, 2022-05-01:
Is it possible to change the function of "5%" and "50%" type buttons from on-off to test button?

I mean, change function from on-off to use it like test button.

Regards.//

abinoer, 2022-05-04:
Sorry, I don't understand what you mean by test button

solawc, 2022-05-21:
Maybe you mean that you want to change the key to, the laser lights up when you press it and turns off when you release it. Is that what you mean? If yes, please reply and confirm. We can evaluate and confirm whether the function needs to be modified.

Zener28, 2022-05-22:
Hello @abinoer & @solawc ..

Yes, @solawc you are right,turn on when you press it and turn off when you release it,

It will be so useful when you are aligning mirrors or you are adjusting something else like focal distance...