makerbase-mks/MKS-DLC32 #198: Yes, we can modify for you. You can send the image.

LuisCnMan, 2022-05-01:
Yes, we can modify for you. You can send the image.

_Originally posted by @makerbase-mks in https://github.com/makerbase-mks/MKS-DLC32/issues/127#issuecomment-1030569749_

LuisCnMan, 2022-05-01:
> Yes, we can modify for you. You can send the image.
> 
> _Originally posted by @makerbase-mks in [#127 (comment)](https://github.com/makerbase-mks/MKS-DLC32/issues/127#issuecomment-1030569749)_

closed 2022-05-01 by LuisCnMan

LuisCnMan, 2022-05-02:
Que características debe tener

LuisCnMan, 2022-05-02:
[image]

me gustaría tener este logo en la pantalla principal

LuisCnMan, 2022-05-02:
me gustaría tener este logo en la pantalla de inicio 

[image]

UlisseDanilo, 2023-02-24:
hi, what is the process to correctly create a custom logo?