makerbase-mks/MKS-DLC32 #199: Enable Hard Homing

Klaus3777, 2022-05-03:
Hello,
with which parameter can you activate the hard homing?
It is disabled on the display.

Greetings Klaus

abinoer, 2022-05-04:
[image]
https://github.com/makerbase-mks/MKS-DLC32/wiki/Firmware
$21=1and$22=1

closed 2022-05-04 by abinoer

Klaus3777, 2022-05-04:
Hello,
this can now be switched on at the display, but after homing the error message "Homig fail, pleas unlock" comes up.

abinoer, 2022-05-04:
Do you have a machine to test? After clicking the button, is the return normal? Can the machine stop when it meets the limit

Klaus3777, 2022-05-04:
this is my test machine, 
after the return and reaching the limit switches comes this message and it stops at X=1 and Y=1

abinoer, 2022-05-04:
Which firmware are you using? Have you tried to slow down the return? You'd better provide a video

Klaus3777, 2022-05-09:
[image]
Hello, 
Update to V2.0.8_H35_20220105_N,
Homing no longer works properly , axes are reversed in the running direction.
Where is the correct home point?
could you set it that way?
if so, how?

abinoer, 2022-05-09:
https://github.com/makerbase-mks/MKS-DLC32/wiki/Firmware
Refer to the instructions and adjust the parameters to change direction. If soft limits and hard limits are on at the same time, zero must be in the lower left corner.